Orchard-Ultrasound-Innovation/TcpInstruments.jl #30: Version 0.4.0

egajda, 2021-03-12:
@JuliaRegistrator register

JuliaRegistrator, 2021-03-12:
Error while trying to register: "Tag with name `v0.3.0` already exists and points to a different commit"

egajda, 2021-03-12:
@JuliaRegistrator register

JuliaRegistrator, 2021-03-12:
Registration pull request created: [JuliaRegistries/General/31816](https://github.com/JuliaRegistries/General/pull/31816)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.0 -m "<description of version>" 73ad60744e932d23a51804d1f53997c76c08ff15
git push origin v0.4.0
```

closed 2021-03-12 by egajda