Orchard-Ultrasound-Innovation/TcpInstruments.jl #37: get_volt_ac not working for Agilent4294a

quintinstedman, 2021-04-07:
When I run this, I get the error message below. get_volt_dc works fine.

ERROR: MethodError: no method matching *(::SubString{String}, ::Unitful.FreeUnits{(V,), 𝐋² 𝐌 𝐈⁻¹ 𝐓⁻³, nothing})
Closest candidates are:
  *(::Any, ::Any, ::Any, ::Any...) at operators.jl:560
  *(::Union{AbstractChar, AbstractString}, ::Union{AbstractChar, AbstractString}...) at strings/basic.jl:260
  *(::Union{Dates.Day, Dates.Hour, Dates.Microsecond, Dates.Millisecond, Dates.Minute, Dates.Nanosecond, Dates.Second, Dates.Week}, ::Unitful.Units) at /Users/admin/.julia/packages/Unitful/PcVKX/src/dates.jl:90
  ...
Stacktrace:
 [1] get_volt_ac(i::TcpInstruments.Instr{Agilent4294A})
   @ TcpInstruments ~/.julia/packages/TcpInstruments/6H4Ex/src/impedance_analyzer/Agilent4294A.jl:57
 [2] top-level scope
   @ REPL[52]:1

mofii, 2021-04-07:
Fixed in commit 89f74f377fec757121c5a98c74190fe8509c11fd
@JuliaRegistrator register branch=main

closed 2021-04-07 by mofii

JuliaRegistrator, 2021-04-07:
Error while trying to register: "Branch `main` not found"

mofii, 2021-04-07:
@JuliaRegistrator register

JuliaRegistrator, 2021-04-07:
Registration pull request created: [JuliaRegistries/General/33785](https://github.com/JuliaRegistries/General/pull/33785)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.6.2 -m "<description of version>" 3ae5c36f86bb577dac2dd4b7c12bc8398aaca874
git push origin v0.6.2
```