Orchard-Ultrasound-Innovation/TcpInstruments.jl #39: Save to Matlab doesn't work for complex numbers

quintinstedman, 2021-04-07:
julia> save(Z, filename="testdata3.mat", format=:matlab)
ERROR: This is the write function for CompositeKind, but the input doesn't fit
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:33
 [2] m_write(mfile::MAT.MAT_HDF5.MatlabHDF5File, parent::HDF5.Group, name::String, s::Quantity{ComplexF64, 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, Unitful.FreeUnits{(Ω,), 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, nothing}})
   @ MAT.MAT_HDF5 ~/.julia/packages/MAT/f523T/src/MAT_HDF5.jl:530
 [3] m_write(mfile::MAT.MAT_HDF5.MatlabHDF5File, parent::HDF5.File, name::String, data::Vector{Quantity{ComplexF64, 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, Unitful.FreeUnits{(Ω,), 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, nothing}}})
   @ MAT.MAT_HDF5 ~/.julia/packages/MAT/f523T/src/MAT_HDF5.jl:479
 [4] write(parent::MAT.MAT_HDF5.MatlabHDF5File, name::String, thing::Vector{Quantity{ComplexF64, 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, Unitful.FreeUnits{(Ω,), 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, nothing}}})
   @ MAT.MAT_HDF5 ~/.julia/packages/MAT/f523T/src/MAT_HDF5.jl:546
 [5] save(data::Vector{Quantity{ComplexF64, 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, Unitful.FreeUnits{(Ω,), 𝐋² 𝐌 𝐈⁻² 𝐓⁻³, nothing}}}; filename::String, format::Symbol)
   @ TcpInstruments ~/.julia/packages/TcpInstruments/6H4Ex/src/instrument.jl:100
 [6] top-level scope
   @ REPL[94]:1

quintinstedman, 2021-04-07:
Actually it appears units are the problem

quintinstedman, 2021-04-07:
Can just strip units before saving using ustrip

mofii, 2021-04-08:
But it's a valid point. I'm opening it again..

mofii, 2021-11-11:
Closed by c0823c08bca7234cabdb1234f35d0dd504c268ff

closed 2021-11-11 by mofii