Orchard-Ultrasound-Innovation/TcpInstruments.jl #43: Add voltage ramping for SRSPS310

mofii, 2021-04-07:


mofii, 2021-04-07:
Fixed in commit 270881f18029b99d5e2b40dfc1dc4d3b41c85e26

mofii, 2021-04-07:
@JuliaRegistrator register

JuliaRegistrator, 2021-04-08:
Registration pull request created: [JuliaRegistries/General/33801](https://github.com/JuliaRegistries/General/pull/33801)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.0 -m "<description of version>" 06ea1a238fedf90ae05f1107fa252a8368447860
git push origin v0.7.0
```

closed 2021-04-08 by mofii