Orchard-Ultrasound-Innovation/TcpInstruments.jl #55: Release v0.8

egajda, 2021-04-09:
@JuliaRegistrator register

JuliaRegistrator, 2021-04-09:
Error while trying to register: "Tag with name `v0.7.1` already exists and points to a different commit"

egajda, 2021-04-09:
@JuliaRegistrator register

JuliaRegistrator, 2021-04-09:
Registration pull request created: [JuliaRegistries/General/33949](https://github.com/JuliaRegistries/General/pull/33949)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.8.0 -m "<description of version>" ab2a3b90e8eefe737da0214f24ea78c05bf5b2f3
git push origin v0.8.0
```

JuliaRegistrator, 2021-04-09:
Registration pull request created: [JuliaRegistries/General/33950](https://github.com/JuliaRegistries/General/pull/33950)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.7.2 -m "<description of version>" de42a5bda17095b74ba59f2e410e3594f293f3a1
git push origin v0.7.2
```

closed 2021-04-12 by mofii