Orchard-Ultrasound-Innovation/TcpInstruments.jl #56: Bump to v0.8.1

mofii, 2021-04-14:


mofii, 2021-04-14:
@JuliaRegistrator register

JuliaRegistrator, 2021-04-14:
Registration pull request created: [JuliaRegistries/General/34255](https://github.com/JuliaRegistries/General/pull/34255)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.8.1 -m "<description of version>" 91c987c8ded3ee51c580de8f5010c9f80881bd62
git push origin v0.8.1
```

JuliaRegistrator, 2021-04-14:
Registration pull request updated: [JuliaRegistries/General/34255](https://github.com/JuliaRegistries/General/pull/34255)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.8.1 -m "<description of version>" 060c7319be594a6be7b554f5e9647c5a1db9308d
git push origin v0.8.1
```

closed 2021-04-14 by mofii