Orchard-Ultrasound-Innovation/TcpInstruments.jl #57: Impedance analyzer units

quintinstedman, 2021-04-29:
The impedance analyzer get_impedance function makes the units of everything ohms. But, data from the impedance analyzer is not always in ohms. It may be degrees.

stevenaleung, 2021-11-08:
Thoughts on making `get_impedance()` output R-X (resistance-reactance) data by default? Degrees can be easily calculated with `angle.(data)`

quintinstedman, 2021-11-08:
That's fine with me as long as the display on the impedance analyzer can still show magnitude and phase. Magnitude and phase is often more useful to view.

stevenaleung, 2021-11-08:
In PR #73, `get_impedance()` switches the impedance analyzer window to complex mode during data transfer then switches back to magnitude & phase mode. We can explore this further to see what works best for you

PR #73 (2021-11-11): Increased code speed for impedance analyzer
Increased code speed for the following:
- Data transfer (240x faster)
- Frequency limit query (20x faster)
- Num data points query (2x faster)

Faster data transfer closes #60 
`get_impedance()` outputting complex impedance closes #57
closed 2021-11-11 by stevenaleung