Orchard-Ultrasound-Innovation/TcpInstruments.jl #59: Complex impedance analyzer data?

quintinstedman, 2021-04-29:
The impedance analyzer data is complex by default, and there is not documentation showing how to make it real. In the typical use case, the data is real.

stevenaleung, 2021-11-08:
@quintinstedman Is this still an existing issue?

Does `real.(ustrip.(data))` solve the problem?

quintinstedman, 2021-11-08:
That does solve the problem. However, if the data is always real (we are not returning impedance as a complex number), we should still update the code to return real numbers instead to simplify things.

stevenaleung, 2021-11-08:
We could have several different functions that each return only one type of measurement. For example, `get_impedance()`, `get_impedance_phase()`, `get_resistance()`, etc. I think this will make it easier to use and debug

Which of the following measurements will be useful?
[image]

quintinstedman, 2021-11-09:
Impedance + phase, real + imaginary, etc. are all mathematically equivalent. I prefer to stick to one so that my data is always in the same format and I avoid confusion. I can then easily calculate the others as needed. I would always just use get_impedance_phase, or get_impedance (if that returns the full complex valued impedance)

I would not use a function like get_resistance that leaves out part of the impedance data.

stevenaleung, 2021-11-09:
Sounds good - let's stick with `get_impedance()` returning full complex valued impedance for now then

closed 2021-11-10 by stevenaleung