Orchard-Ultrasound-Innovation/TcpInstruments.jl #60: Agilent 4294a timeout

quintinstedman, 2021-04-29:
Depending on the BW setting and number of points, it can take a substantial amount of time for the Agilent4294a to get all the data (a few seconds or more). get_impedance times out if it takes to long to get the data. This limits the settings we can use. It should not time out so quickly.

mofii, 2021-11-01:
We should make a proper fix for this that doesn't induce such a high timeout / slow read rate.

stevenaleung, 2021-11-02:
These are the two relevant sections of code:
https://github.com/Orchard-Ultrasound-Innovation/TcpInstruments.jl/blob/c4d59aa2c0db51ced7c58c24d68b15e2d8182043/src/impedance_analyzer/Agilent4294A.jl#L67-L78

https://github.com/Orchard-Ultrasound-Innovation/TcpInstruments.jl/blob/c4d59aa2c0db51ced7c58c24d68b15e2d8182043/src/instrument.jl#L217-L235

The 20 second timeout set in line 68 shouldn't be making things unnecessarily slow since line 226 breaks out of the while loop once the Task `proc` has completed

To speed up the code, perhaps a longer polling interval like 0.1 sec could be used instead. Some refactoring could be done using the `timedwait()` function

Lines 70-76 could also be replaced with a `reshape()` call

commit fd3c5a41 (2021-04-29): Increase timeout to 20s for Agilent4294a. Closes #60
PR #73 (2021-11-11): Increased code speed for impedance analyzer
Increased code speed for the following:
- Data transfer (240x faster)
- Frequency limit query (20x faster)
- Num data points query (2x faster)

Faster data transfer closes #60 
`get_impedance()` outputting complex impedance closes #57
closed 2021-11-11 by stevenaleung