Orchard-Ultrasound-Innovation/TcpInstruments.jl #61: Bump to v 0.8.2

mofii, 2021-04-29:


mofii, 2021-04-29:
@JuliaRegistrator register

JuliaRegistrator, 2021-04-29:
Registration pull request created: [JuliaRegistries/General/35687](https://github.com/JuliaRegistries/General/pull/35687)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.8.2 -m "<description of version>" bc8de8be40818a2608b15fa0e380726efb3dfcf0
git push origin v0.8.2
```

closed 2021-04-30 by mofii