Orchard-Ultrasound-Innovation/TcpInstruments.jl #71: Scope grab data bug

quintinstedman, 2021-09-23:
I am using 3 channels on the scope. When I run get_data(scope, [1,2,3]) the scope stops running (run/stop light in the top right corner turns red), and it hangs and never return data. Sometimes it returns an error instead.

Error Message:
ERROR: IOError: read: connection reset by peer (ECONNRESET)
Stacktrace:
  [1] wait_readnb(x::Sockets.TCPSocket, nb::Int64)
    @ Base ./stream.jl:408
  [2] (::Base.var"#wait_locked#595")(s::Sockets.TCPSocket, buf::IOBuffer, nb::Int64)
    @ Base ./stream.jl:848
  [3] readbytes!(s::Sockets.TCPSocket, a::Vector{UInt8}, nb::Int64)
    @ Base ./stream.jl:854
  [4] read(s::Sockets.TCPSocket, nb::Int64)
    @ Base ./io.jl:962
  [5] scope_read_raw_waveform(instr::TcpInstruments.Instr{AgilentDSOX4034A})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/rqBHX/src/oscilloscope/scope_common.jl:249
  [6] get_data(instr::TcpInstruments.Instr{AgilentDSOX4034A}, ch::Int64; scope_stats::Bool)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/rqBHX/src/oscilloscope/scope_common.jl:259
  [7] #57
    @ ./none:0 [inlined]
  [8] iterate
    @ ./generator.jl:47 [inlined]
  [9] collect_to!
    @ ./array.jl:724 [inlined]
 [10] collect_to_with_first!(dest::Vector{TcpInstruments.ScopeData}, v1::TcpInstruments.ScopeData, itr::Base.Generator{Vector{Int64}, TcpInstruments.var"#57#61"{Bool, TcpInstruments.Instr{AgilentDSOX4034A}}}, st::Int64)
    @ Base ./array.jl:702
 [11] collect(itr::Base.Generator{Vector{Int64}, TcpInstruments.var"#57#61"{Bool, TcpInstruments.Instr{AgilentDSOX4034A}}})
    @ Base ./array.jl:683
 [12] get_data(instr::TcpInstruments.Instr{AgilentDSOX4034A}, ch_vec::Vector{Int64}; inbounds::Bool, scope_stats::Bool)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/rqBHX/src/oscilloscope/scope_common.jl:286
 [13] get_data(instr::TcpInstruments.Instr{AgilentDSOX4034A}, ch_vec::Vector{Int64})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/rqBHX/src/oscilloscope/scope_common.jl:267
 [14] top-level scope
    @ REPL[41]:1

Grabbing data front single channels works For example, 
get_data(scope, 1)
get_data(scope, 2)
get_data(scope, 3)

If you get data from a single channel first then run get_data(scope, [1,2,3]) it works fine.

quintinstedman, 2021-09-23:
For it to work properly, you need to call get_data for each channel individually before you can call it as part of a group.

If you do
test1= get_data(scope, 1);
test2= get_data(scope, 2);
test123= get_data(scope, [1,2,3]);
It will fail

It works if you do
test1= get_data(scope, 1);
test2= get_data(scope, 2);
test3= get_data(scope, 2);
test123= get_data(scope, [1,2,3]);

stevenaleung, 2021-11-11:
I wasn't able to recreate this bug with the `AgilentDSOX4034A` using release v0.8.4

@quintinstedman Let's try to do it on your machine

mofii, 2021-12-01:
@quintinstedman did you try this again?

quintinstedman, 2021-12-01:
I don't get the problem anymore using the latest version.

closed 2021-12-01 by quintinstedman

mofii, 2021-12-01:
Awesome!