Orchard-Ultrasound-Innovation/TcpInstruments.jl #83: Save to Matlab doesn't work anymore.

quintinstedman, 2021-12-07:
The save function used to work, but does not anymore.

For example, if I try

a = 1;
save(V_bias_meas, filename= "test", format=:matlab)

I get an an error
RROR: MethodError: no method matching raw(::Float64)
Closest candidates are:
  raw(::Union{Quantity{T, 𝐈, U}, Level{L, S, Quantity{T, 𝐈, U}} where {L, S}} where {T, U}) at /Users/admin/.julia/packages/TcpInstruments/TgF34/src/util.jl:14
  raw(::Union{Quantity{T, 𝐋² 𝐌 𝐈⁻¹ 𝐓⁻³, U}, Level{L, S, Quantity{T, 𝐋² 𝐌 𝐈⁻¹ 𝐓⁻³, U}} where {L, S}} where {T, U}) at /Users/admin/.julia/packages/TcpInstruments/TgF34/src/util.jl:15
  raw(::Union{Quantity{T, 𝐓⁻¹, U}, Level{L, S, Quantity{T, 𝐓⁻¹, U}} where {L, S}} where {T, U}) at /Users/admin/.julia/packages/TcpInstruments/TgF34/src/util.jl:16
  ...
Stacktrace:
 [1] _broadcast_getindex_evalf
   @ ./broadcast.jl:648 [inlined]
 [2] _broadcast_getindex
   @ ./broadcast.jl:621 [inlined]
 [3] getindex
   @ ./broadcast.jl:575 [inlined]
 [4] copy
   @ ./broadcast.jl:922 [inlined]
 [5] materialize
   @ ./broadcast.jl:883 [inlined]
 [6] save(data::Matrix{Float64}; filename::String, format::Symbol)
   @ TcpInstruments ~/.julia/packages/TcpInstruments/TgF34/src/util.jl:56
 [7] top-level scope
 
 If I do not put format=:matlab, then it still works.

stevenaleung, 2021-12-07:
The problem is that the save function is expecting Unitful values. Does the following throw errors?
```
val = 1u"V";
save(val, filename="test", format=:matlab)
```

quintinstedman, 2021-12-07:
That works, but I'm sure I could save unitless quantities before. Plenty of data we might want to record from a measurement doesn't have units, so it doesn't really make sense to require them.

mofii, 2021-12-07:
Out of curiosity, @quintinstedman where are you getting the unitless numbers from?
I think ideally we should have all instruments use numbers with units.

mofii, 2021-12-07:
For now, you can try this branch and see if that works for you:
```julia
> using Pkg; Pkg.add(name="TcpInstruments", rev="ustrip_saving")
> using TcpInstruments
> val = 1
> save(val, filename="test", format=:matlab)
```

mofii, 2021-12-07:
@stevenaleung what do you think, should we merge that branch into main?
The downside is that it doesn't do what raw() does: makes sure the scale is the base of the unit (e.g., volts instead of millivolts) before stripping the unit.
Writing this out, I think the way to go is to change raw() to have a fall-back and return the input unaltered if it has no unit.

PR #84 (2021-12-07): Fix save to MAT file
Closes #83
closed 2021-12-07 by stevenaleung