Orchard-Ultrasound-Innovation/TcpInstruments.jl #101: get_impedance cannot be run consecutively

juliettefowler, 2021-12-21:
### Bug description
get_impedance cannot be run more than one time without a pause or delay. 

### Steps or code to reproduce the behavior
```
using TcpInstruments

ia = initialize(Agilent4294A, "10.1.30.39")
for j = 1:2
     data = get_impedance(ia)
end
...
```

### Versions
TcpInstruments.jl version/commit: v0.9.0 
Julia version: 1.6.3

stevenaleung, 2021-12-21:
@juliettefowler Could you add the error messages/stack to the "Steps or code to reproduce the behavior" section? I'll update the template to ask for that

juliettefowler, 2021-12-22:
ERROR: Query timed out
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:33
 [2] query(instr::TcpInstruments.Instr{Agilent4294A}, message::String; timeout::Float64)
   @ TcpInstruments ~/.julia/packages/TcpInstruments/PMJpp/src/instrument.jl:241
 [3] f_query(obj::TcpInstruments.Instr{Agilent4294A}, ins::String; timeout::Float64)
   @ TcpInstruments ~/.julia/packages/TcpInstruments/PMJpp/src/instrument.jl:249
 [4] f_query
   @ ~/.julia/packages/TcpInstruments/PMJpp/src/instrument.jl:249 [inlined]
 [5] get_volt_dc
   @ ~/.julia/packages/TcpInstruments/PMJpp/src/impedance_analyzer/all.jl:137 [inlined]
 [6] get_impedance_analyzer_info(ia::TcpInstruments.Instr{Agilent4294A})
   @ TcpInstruments ~/.julia/packages/TcpInstruments/PMJpp/src/impedance_analyzer/all.jl:6
 [7] get_impedance(ia::TcpInstruments.Instr{Agilent4294A})
   @ TcpInstruments ~/.julia/packages/TcpInstruments/PMJpp/src/impedance_analyzer/Agilent4294A.jl:68
 [8] top-level scope
   @ ./REPL[15]:2

PR #121 (2022-04-22): Fixed errors due to query timeouts
Closes #101
Closes #102 
Closes #103 

Alternatively, `get_volt_dc()` and `get_volt_ac()` could've been refactored to use `f_query(; timeout=0)`. The current change was made to maintain consistency with the rest of the impedance analyzer `get` functions
closed 2022-04-22 by stevenaleung