Orchard-Ultrasound-Innovation/TcpInstruments.jl #103: Sleep required after set voltage

quintinstedman, 2022-01-15:
### Bug description
A pause is required between using set_voltage on the PS310 power supply and using another instrument control command. A 1 second pause after setting the voltage prevents the error. 0.5 seconds is not usually enough. The issue does not exist if you repeatedly call set_voltage for the ps310. You get it if, for example, you set the voltage and then use get_impedance. Likeley this issue may exist in other situations where commands are send to different pieces of equipment in quick succession.

### Steps or code to reproduce the behavior

using TcpInstruments
using Unitful
Z_analyzer= initialize(Agilent4294A,"10.1.30.39");
ps = initialize(SRSPS310, "10.1.30.37:1234"; GPIB_ID=2)
set_voltage(ps,10u"V");
sleep(0.1)
Z_data= get_impedance(Z_analyzer);

Quick Fix:
using TcpInstruments
using Unitful
Z_analyzer= initialize(Agilent4294A,"10.1.30.39");
ps = initialize(SRSPS310, "10.1.30.37:1234"; GPIB_ID=2)
set_voltage(ps,10u"V");
sleep(1)
Z_data= get_impedance(Z_analyzer);

```

### Versions
TcpInstruments.jl version/commit: 0.9.0
Julia version: 1.6.1

stevenaleung, 2022-01-15:
Interesting.. could you also provide the error message stack? (We'll modify the template to ask for it by default)

quintinstedman, 2022-01-15:
ERROR: LoadError: Query timed out
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:33
 [2] query(instr::TcpInstruments.Instr{Agilent4294A}, message::String; timeout::Float64)
   @ TcpInstruments ~/.julia/packages/TcpInstruments/K2yi5/src/instrument.jl:234
 [3] f_query(obj::TcpInstruments.Instr{Agilent4294A}, ins::String; timeout::Float64)
   @ TcpInstruments ~/.julia/packages/TcpInstruments/K2yi5/src/instrument.jl:242
 [4] f_query
   @ ~/.julia/packages/TcpInstruments/K2yi5/src/instrument.jl:242 [inlined]
 [5] get_volt_dc
   @ ~/.julia/packages/TcpInstruments/K2yi5/src/impedance_analyzer/all.jl:182 [inlined]
 [6] get_impedance_analyzer_info(ia::TcpInstruments.Instr{Agilent4294A})
   @ TcpInstruments ~/.julia/packages/TcpInstruments/K2yi5/src/impedance_analyzer/all.jl:18
 [7] get_impedance(ia::TcpInstruments.Instr{Agilent4294A})
   @ TcpInstruments ~/.julia/packages/TcpInstruments/K2yi5/src/impedance_analyzer/Agilent4294A.jl:68
 [8] top-level scope
   @ /Volumes/GoogleDrive/Shared drives/user-Quintin/Julia_Functions/Impedance_bias_sweep.jl:51
in expression starting at /Volumes/GoogleDrive/Shared drives/user-Quintin/Julia_Functions/Impedance_bias_sweep.jl:41

stevenaleung, 2022-01-15:
Took a quick look at the code but I haven't found areas where the PS310 could be blocking the impedance analyzer yet. Is this behavior reproduced when you put the sleep command between the initialization and function calls?
```
using TcpInstruments
using Unitful
Z_analyzer= initialize(Agilent4294A, "10.1.30.39");
ps = initialize(SRSPS310, "10.1.30.37:1234"; GPIB_ID=2);
sleep(1);
set_voltage(ps, 10u"V");
Z_data= get_impedance(Z_analyzer);
```

quintinstedman, 2022-01-15:
It's getting confusing now. Even just running

`Z_analyzer= initialize(Agilent4294A, "10.1.30.39");`

gives me an error now.

ERROR: IOError: connect: connection timed out (ETIMEDOUT)
Stacktrace:
 [1] wait_connected(x::Sockets.TCPSocket)
   @ Sockets /Users/julia/buildbot/worker/package_macos64/build/usr/share/julia/stdlib/v1.6/Sockets/src/Sockets.jl:532
 [2] connect
   @ /Users/julia/buildbot/worker/package_macos64/build/usr/share/julia/stdlib/v1.6/Sockets/src/Sockets.jl:567 [inlined]
 [3] connect(host::SubString{String}, port::Int64)
   @ Sockets /Users/julia/buildbot/worker/package_macos64/build/usr/share/julia/stdlib/v1.6/Sockets/src/Sockets.jl:552
 [4] connect!(instr::TcpInstruments.Instr{Agilent4294A})
   @ TcpInstruments ~/.julia/packages/TcpInstruments/K2yi5/src/instrument.jl:186
 [5] #initialize#3
   @ ~/.julia/packages/TcpInstruments/K2yi5/src/instrument.jl:110 [inlined]
 [6] initialize(model::Type{Agilent4294A}, address::String)
   @ TcpInstruments ~/.julia/packages/TcpInstruments/K2yi5/src/instrument.jl:109
 [7] top-level scope
   @ REPL[32]:1

Need to try restarting everything.

quintinstedman, 2022-01-24:
We've found some more cases of this bug. I will put together a more thorough description of cases where it arises and example code to reproduce the issue. It's ok to leave it for now. We can solve the problem by adding pauses to the code.

PR #121 (2022-04-22): Fixed errors due to query timeouts
Closes #101
Closes #102 
Closes #103 

Alternatively, `get_volt_dc()` and `get_volt_ac()` could've been refactored to use `f_query(; timeout=0)`. The current change was made to maintain consistency with the rest of the impedance analyzer `get` functions
closed 2022-04-22 by stevenaleung