Orchard-Ultrasound-Innovation/TcpInstruments.jl #105: Make brand name / maker name instrument types

mofii, 2022-01-25:
Let  let each device inherit from their corresponding make type.
Let all commands dispatch on the make type instead of the specific model type.
This would allow one to dispatch on the brand name, which could be useful when using a device that hasn't been tested/added yet. This assumes that devices made by the same maker will act similarly. 
For devices from the same maker which act differently, maybe that's only the case for a subset of the supported commands, and for this subset a new method that dispatches directly on the model number should be defined.

stevenaleung, 2022-03-22:
Looks like a duplicate of #96

mofii, 2022-03-22:
I agree. I'm closing this one.

closed 2022-03-22 by mofii