Orchard-Ultrasound-Innovation/TcpInstruments.jl #108: Setting frequency on waveform generator does not work

quintinstedman, 2022-03-21:
### Bug description
Setting the frequency on the 33612A waveform generators does not work. The signal generator gives an error "Error generated by remote interface" on its display.

### Steps or code to reproduce the behavior
```
using TcpInstruments
...
data_path= "//Volumes/GoogleDrive/Shared drives/measurements/CMUT_characterizations/manufacturing_SNF001/assembly_STR001/design_200x4400/array_004/frequency_sweep_single_element_2022-03-21/";
base_file_name= "SNF001-STR001-200x4400-T3";

freq= collect(2:1:5)u"MHz";

scope = initialize(AgilentDSOX4034A,"10.1.30.32");
sleep(0.5);
sg = initialize(Keysight33612A, "10.1.30.47")
sleep(0.5);

for i=1:length(freq)
    set_frequency(sg, freq[i])
    sleep(0.5);
    scope_data= get_data(scope, [1,2,3])
    t= ustrip.(scope_data[1].time);
    V_trig= ustrip.(scope_data[1].volt);
    V_TX= ustrip.(scope_data[3].volt);
    V_hydrophone= ustrip.(scope_data[2].volt);
    n_points= length(t);
    signal_array= zeros(n_points,4)
    signal_array[:,1]= t;
    signal_array[:,2]= V_trig;
    signal_array[:,3]= V_TX;
    signal_array[:,4]= V_hydrophone;

    data_filename= string(data_path, base_file_name, "-", ustrip(freq[i]).*1e-6, "MHz");
    #save(signal_array, filename=data_filename, format=:matlab);
    sleep(0.1)
end

### Versions
TcpInstruments.jl version/commit: v0.8.7
Julia version: 1.6.1

stevenaleung, 2022-03-22:
Is this blocking your experiments? How urgent is the bug fix? I can look into it as soon as tomorrow

quintinstedman, 2022-03-22:
I stepped the frequency manually. It's not now, so not as urgent. Sometime this week would be good, but doesn't have to be tomorrow.

PR #109 (2022-03-22): Resolved signal generator set_frequency() bug
Closes #108
closed 2022-03-22 by stevenaleung