Orchard-Ultrasound-Innovation/TcpInstruments.jl #125: Cannount load personal YAML file in Pluto.jl environment

wiseac, 2022-06-01:
Hello,

I am using Pluto.jl to run TcpInstruments.jl and its great except for this nifty feature Pluto.jl does. It creates its own package environment separate from the Hosts downloaded packages. This is really useful for sharing notebooks to other people without needing them to download every package but it also limits how much you can edit the package. My problem lies with when trying to load a new instruments config file it doesn't load/recognize the working directory where im trying to define the location of the file and instead says "Cannot load file". 

I tried using instrumentConfig.jl to define a Github path and it still did not work. I was wondering if there was a way define the instrument config file as one in the current workspace or something along those lines.

If not it seems more like a Pluto issue.

Thanks,
Wise

wiseac, 2022-06-01:
Hello,

[image]

This is what I am getting.

mofii, 2022-06-01:
It's telling you that the device name `SDS1204XE` is not a valid name because it has not been implemented.

mofii, 2022-06-01:
I see that it's a 4-channel scope by SIGLENT.
I don't know how much their implementation of SCPI differs from that of Agilent. You could test your luck by just telling the code that it's an Agilent scope.

wiseac, 2022-06-01:
Hello, 

Thats what i have been doing and just changing the IP as needed. We are planning on adding Arduinos and RasberryPi's soon so it would be great if could actually name the models we are using.

Thanks,

mofii, 2022-06-01:
We would welcome a PR to implement support SIGLENT scopes.
It should be pretty straight forward.
You need to make a new instrument type in oscilloscopes/types.jl (the last two lines are the new additions):
```Julia
abstract type AgilentScope <: Oscilloscope end
struct AgilentDSOX4024A <: AgilentScope end
struct AgilentDSOX4034A <: AgilentScope end

abstract type SiglentScope <: Oscilloscope end
struct SDS1204XE <: SiglentScope end
```

Then copy oscilloscopes/agilent_common.jl to oscilloscopes/siglent_common.jl, do a search and replace on `AgilentScope` to `SiglentScope` and then adjust the functions that needs to be adjusted.