Orchard-Ultrasound-Innovation/TcpInstruments.jl #127: Feature request: arbitrary SCPI commands

quintinstedman, 2022-06-03:
Many lab instruments have hundreds of commands available, and it's not reasonable to expect TcpInstruments to have functionality covering all of them. It would be helpful if the user could send arbitrary SCPI commands to an instrument and read the responses to queries. This would allow more advanced users to use the more specialized capabilities of the instruments while keeping the package streamlined.

mofii, 2022-06-08:
It's undocumented, but the internal function that sends the commands to the devices is `f_query()`
You could try:
```julia
TcpInstruments.f_query(instr, "<SCPI cmd string>"; timeout=3)
```

mofii, 2022-06-10:
@quintinstedman see PR #130 that I expect will be included in the next release.

PR #130 (2022-06-12): Custom_scpi_cmd
Better description of how to send custom commands to devices.
This closes #127
commit 00b52ab9 (2022-06-12): Merge pull request #130 from Orchard-Ultrasound-Innovation:custom_scpi_cmd

Custom_scpi_cmd
Closes #127
closed 2022-06-12 by mofii