Orchard-Ultrasound-Innovation/TcpInstruments.jl #191: get_data(scope) crashes at seemingly random times

mofii, 2022-12-13:
### Bug description
Veijo and Quintin were sweeping the DC voltage and repeatedly grabbing data from an oscilloscope.

### Steps to reproduce the behavior along with the error stacktrace
```
TBD - I'll ask for the code they were running.

```

This is the stack trace:
```julia
ERROR: TaskFailedException
Stacktrace:
  [1] wait
    @ ./task.jl:334 [inlined]
  [2] read_with_timeout(instr::TcpInstruments.Instrument{AgilentDSOX4034A}, timeout_sec::Float64)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/instrument.jl:335
  [3] #query#4
    @ ~/.julia/packages/TcpInstruments/Q778A/src/instrument.jl:258 [inlined]
  [4] i_query(instr::TcpInstruments.Instrument{AgilentDSOX4034A}, message::String; timeout::Float64)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/instrument.jl:305
  [5] i_query
    @ ~/.julia/packages/TcpInstruments/Q778A/src/instrument.jl:305 [inlined]
  [6] get_num_data_points
    @ ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:174 [inlined]
  [7] read_uint16(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:105
  [8] read_raw_waveform(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:73
  [9] get_data(instr::TcpInstruments.Instrument{AgilentDSOX4034A}, ch::Int64)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:59
 [10] #56
    @ ./none:0 [inlined]
 [11] iterate
    @ ./generator.jl:47 [inlined]
 [12] collect(itr::Base.Generator{Vector{Int64}, TcpInstruments.var"#56#57"{TcpInstruments.Instrument{AgilentDSOX4034A}}})
    @ Base ./array.jl:724
 [13] get_data(instr::TcpInstruments.Instrument{AgilentDSOX4034A}, ch_vec::Vector{Int64}; check_channels::Bool)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:52
 [14] get_data
    @ ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:42 [inlined]
 [15] grabHydrophoneData(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ Main.FatigueTestUtilities ~/Documents/Github/Lab-Measurement-Tools/FatigueTestUtilities.jl:243
 [16] chargingCheck(smu::TcpInstruments.Instrument{AgilentDSOX4024A}, scope::TcpInstruments.Instrument{AgilentDSOX4034A}, waveform_gen::TcpInstruments.Instrument{Keysight33612A}, V_waveform_gen::Float64, V_dc::Vector{Float64})
    @ Main.FatigueTestUtilities ~/Documents/Github/Lab-Measurement-Tools/FatigueTestUtilities.jl:198
 [17] top-level scope
    @ ~/Documents/Github/Lab-Measurement-Tools/run_single_element_fatigue_test.jl:113

    nested task error: InterruptException:
    Stacktrace:
     [1] try_yieldto(undo::typeof(Base.ensure_rescheduled))
       @ Base ./task.jl:812
     [2] wait()
       @ Base ./task.jl:872
     [3] wait(c::Base.GenericCondition{Base.Threads.SpinLock})
       @ Base ./condition.jl:123
     [4] readuntil(x::Sockets.TCPSocket, c::UInt8; keep::Bool)
       @ Base ./stream.jl:958
     [5] readline(s::Sockets.TCPSocket; keep::Bool)
       @ Base ./io.jl:509
     [6] readline
       @ ./io.jl:509 [inlined]
     [7] read(instr::TcpInstruments.Instrument{AgilentDSOX4034A})
       @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/instrument.jl:229
     [8] macro expansion
       @ ~/.julia/packages/TcpInstruments/Q778A/src/instrument.jl:331 [inlined]
     [9] (::TcpInstruments.var"#7#8"{TcpInstruments.Instrument{AgilentDSOX4034A}, Float64, Channel{Any}})()
       @ TcpInstruments ./task.jl:429
```
### Versions
TcpInstruments.jl version: 0.11.8
Julia version: 1.7.3

### Priority level (optional)
Highest

OUI-lab-equipment, 2023-01-10:
Experienced another crash while sweeping frequency:
```
ERROR: InterruptException:
Stacktrace:
  [1] try_yieldto(undo::typeof(Base.ensure_rescheduled))
    @ Base ./task.jl:812
  [2] wait()
    @ Base ./task.jl:872
  [3] wait(c::Base.GenericCondition{Base.Threads.SpinLock})
    @ Base ./condition.jl:123
  [4] wait_readnb(x::Sockets.TCPSocket, nb::Int64)
    @ Base ./stream.jl:413
  [5] (::Base.var"#wait_locked#647")(s::Sockets.TCPSocket, buf::IOBuffer, nb::Int64)
    @ Base ./stream.jl:848
  [6] readbytes!(s::Sockets.TCPSocket, a::Vector{UInt8}, nb::Int64)
    @ Base ./stream.jl:854
  [7] read(s::Sockets.TCPSocket, nb::Int64)
    @ Base ./io.jl:965
  [8] get_data_header(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:149
  [9] get_num_data_bytes(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:137
 [10] read_uint8(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:83
 [11] read_raw_waveform(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:71
 [12] get_data(instr::TcpInstruments.Instrument{AgilentDSOX4034A}, ch::Int64)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:59
 [13] #56
    @ ./none:0 [inlined]
 [14] iterate
    @ ./generator.jl:47 [inlined]
 [15] collect(itr::Base.Generator{Vector{Int64}, TcpInstruments.var"#56#57"{TcpInstruments.Instrument{AgilentDSOX4034A}}})
    @ Base ./array.jl:724
 [16] get_data(instr::TcpInstruments.Instrument{AgilentDSOX4034A}, ch_vec::Vector{Int64}; check_channels::Bool)
    @ TcpInstruments ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:52
 [17] get_data
    @ ~/.julia/packages/TcpInstruments/Q778A/src/oscilloscope/agilent_common.jl:42 [inlined]
 [18] grabHydrophoneData(scope::TcpInstruments.Instrument{AgilentDSOX4034A})
    @ Main.FatigueTestUtilities ~/Documents/Github/Lab-Measurement-Tools/FatigueTestUtilities.jl:243
 [19] top-level scope
    @ ~/Documents/Github/Lab-Measurement-Tools/run_frequency_sweep_measurement.jl:130
```

mofii, 2023-01-10:
@quintinstedman @OUI-lab-equipment 
Thanks! 
This actually looks like a different issue. From the interrupt exception it looks like the code hung and you hit `ctrl-c` to exit.
This seems to be issue #215.

The good news is that we have code ready for #215 and the main branch already has a merged PR that probably fixes this issue (#191).

We'll release a new version later this week.

mofii, 2023-01-10:
This bug is probably squashed by PR #192.